m5stack/M5Dial #1: Compilation error: exit status 1

stephenhouser, 2023-11-16:
Arduino IDE v2.2.1
M5Stack Boards v2.0.8
M5Dial Library v1.01

```
#include "M5Dial.h"

void setup() {
}

void loop() {
}
```

Error Log:
```
FQBN: esp32:esp32:stamp-s3
Using board 'stamp-s3' from platform in folder: /Users/hidden/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11
Using core 'esp32' from platform in folder: /Users/hidden/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11

bash -c "[ ! -f \"/private/var/folders/s5/9fwg2bbd5sggdqs1hlbp2pp00000gn/T/.arduinoIDE-unsaved20231016-14463-dxw3n8.09lmu/encoder\"/partitions.csv ] || cp -f \"/private/var/folders/s5/9fwg2bbd5sggdqs1hlbp2pp00000gn/T/.arduinoIDE-unsaved20231016-14463-dxw3n8.09lmu/encoder\"/partitions.csv \"/private/var/folders/s5/9fwg2bbd5sggdqs1hlbp2pp00000gn/T/arduino/sketches/CEE4FA9D792111B14A5E963407FC845A\"/partitions.csv"

--- removed for brevity ---

@/private/var/folders/s5/9fwg2bbd5sggdqs1hlbp2pp00000gn/T/arduino/sketches/CEE4FA9D792111B14A5E963407FC845A/file_opts -I/Users/hidden/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/cores/esp32 -I/Users/hidden/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/variants/m5stack_stamp_s3 -I/Users/hidden/Documents/Arduino/libraries/M5Dial/src -I/Users/hidden/Documents/Arduino/libraries/M5Unified/src -I/Users/hidden/Documents/Arduino/libraries/M5GFX/src -I/Users/hidden/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/Wire/src /Users/hidden/Documents/Arduino/libraries/M5Dial/src/utility/MFRC522.cpp -o /private/var/folders/s5/9fwg2bbd5sggdqs1hlbp2pp00000gn/T/arduino/sketches/CEE4FA9D792111B14A5E963407FC845A/libraries/M5Dial/utility/MFRC522.cpp.o
Using previously compiled file: /private/var/folders/s5/9fwg2bbd5sggdqs1hlbp2pp00000gn/T/arduino/sketches/CEE4FA9D792111B14A5E963407FC845A/libraries/M5Dial/M5Dial.cpp.o
/Users/hidden/Documents/Arduino/libraries/M5Dial/src/utility/MFRC522.cpp: In member function 'uint8_t MFRC522::PCD_CommunicateWithPICC(uint8_t, uint8_t, uint8_t*, uint8_t, uint8_t*, uint8_t*, uint8_t*, uint8_t, bool)':
/Users/hidden/Documents/Arduino/libraries/M5Dial/src/utility/MFRC522.cpp:464:20: error: '_validBits' may be used uninitialized in this function [-Werror=maybe-uninitialized]
   if (*backLen < 2 || _validBits != 0) {
       ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~
cc1plus: some warnings being treated as errors

Using library M5Dial at version 1.0.1 in folder: /Users/hidden/Documents/Arduino/libraries/M5Dial 
Using library M5Unified at version 0.1.10 in folder: /Users/hidden/Documents/Arduino/libraries/M5Unified 
Using library M5GFX at version 0.1.11 in folder: /Users/hidden/Documents/Arduino/libraries/M5GFX 
Using library Wire at version 2.0.0 in folder: /Users/hidden/Library/Arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/Wire 
exit status 1

Compilation error: exit status 1
```

stephenhouser, 2023-11-16:
NOTE: Tried this on a macOS system and an Arch Linux system with the same result. Both, new clean installs of Arduino IDE, boards, and libraries.

Gitshaoxiang, 2023-11-21:
hi. thanks for your feedback, now we have released a new version.  you could test it later

closed 2023-11-29 by Gitshaoxiang