m5stack/M5Dial #5: Touch stops working.

bdring, 2023-12-06:
Sometimes my touch screen stops working and it requires a power cycle to fix.

Is there anything that can be done in code to detect and fix the problem?

bdring, 2023-12-08:
This is the project I am working on. It works great 99% of the time, but sometimes the touch quits.

https://www.youtube.com/watch?v=mg1y1k8I-8s

felmue, 2023-12-11:
Hello @bdring

I gather touch stops working randomly? On the Wiki page, section Power, you mention that currently USB is used to supply power to the M5Dial. I was wondering if the touch maybe stops working due to some power glitch? Have you tried an alternate power source?

BTW: very nice project, video and documentation.

Thanks
Felix

chuckgiddens, 2025-08-17:
I have same issue. I am running on lipo battery and after poweroff the touch screen no longer works almost every time