m5stack/M5Dial #11: Cannot wakeup from deep sleep with WAKE button

MitchBradley, 2024-01-10:
The documentation and schematics say that GPIO 42 - the dial switch - is for WAKE.  When I put the ESP32-S3 into deep sleep mode, I cannot get it to wake up by pressing the dial switch.

The Espressif documents say that only RTC GPIOs can be used for wakeup from deep sleep.  According to the ESP32-S3 datasheet at https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf , only GPIOS 0-21 are RTC GPIOs.

Is there any way to wake from deep sleep with a button press on the M5 Dial?

felmue, 2024-01-11:
Hello @MitchBradley 

the dial switch cannot be used to wakeup ESP32 from light or deep sleep, for the reason you already mentioned above - GPIO42 is not an RTC GPIO.

What the dial switch can do is wake the system from a full shutdown state.

Thanks
Felix

MitchBradley, 2024-01-11:
What do you mean by full shutdown?  How does one enter "full shutdown" ?

felmue, 2024-01-11:
Hello @MitchBradley 

please have a look at the wakeup example [here](https://github.com/m5stack/M5Dial/blob/master/examples/Basic/wakeup/wakeup.ino).

Below line (commented out in the example) does a full shutdown.
```
M5Dial.Power.powerOff();
```
Note: in full shutdown state only the RTC is still powered which can also be used to wake up after some time.

Thanks
Felix

jsgagnon2020, 2024-02-07:
Hello,
Based on the previous comment and my tests, 
it is possible to go in "sleep mode" /or/ "shutdown mode" by coding uniquely.  (the example does that)

But
Is it possible to get out of these two modes uniquely base on the "rotating knob" /or/ "push button" /or/  "touch screen" ?

Meaning: Waking up from sleep or shutdown the M5Dial uniquely with the "rotating knob" /or/ "push button" /or/  "touch screen" ?
att.: Uniquely and NOT through a pin state? 

Thank, JS

felmue, 2024-02-08:
Hello @jsgagnon2020 

getting out of shutdown mode is only possible via "push button" (WAKE) or RTC (INT) which are connected to the wake-up circuit. See schematic [here](https://docs.m5stack.com/en/core/M5Dial). ("rotating knob" and "touch screen" are **not** connected to the wake-up circuit.) 

Getting out of sleep mode is only possible via "touch screen" (TP_INT) which uses an RTC GPIO. ("rotating knob" and "push button" cannot be used for that as they are connected to **non** RTC GPIOs.)

You can use the wake-up cause to figure out what has woken up the system. See [here](https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32s3/api-reference/system/sleep_modes.html#_CPPv418esp_sleep_source_t).

Thanks
Felix

jsgagnon2020, 2024-02-08:
:) :) most appreciated, when I will have the Arduino code for the M5Dial, I will share it

felmue, 2024-02-09:
Hello @jsgagnon2020 

I've created a simple example show-casting sleep modes and waking from touch screen or dial button.

Thanks
Felix

jsgagnon2020, 2024-02-09:
Mostly appreciated! 😊

Envoyé à partir de Courrier<https://go.microsoft.com/fwlink/?LinkId=550986> pour Windows

jsgagnon2020, 2024-02-09:
After testing,

I found out that only the touchPad work for the deepsleep

Thank to all

Here is my code for the M5Dial

//--------------

#include <Arduino.h>
#include "M5Dial.h"
RTC_DATA_ATTR  int   mode   = 0;

#define uS_TO_S_FACTOR 1000000
#define TIME_TO_SLEEP  5

unsigned long last_interaction = 0;

void setup()
            {
              auto cfg = M5.config();  M5Dial.begin(cfg, true, false);

              M5Dial.Display.setTextColor(GREEN);  M5Dial.Display.setTextDatum(middle_center);  M5Dial.Display.setTextFont( &fonts::FreeSerifItalic18pt7b);  M5Dial.Display.setTextSize( 1);
              M5Dial.Display.drawString("Sleep tests",        M5Dial.Display.width() / 2,           M5Dial.Display.height() / 2 - 70);  delay(2000);

              esp_sleep_wakeup_cause_t wakeup_reason;         wakeup_reason = esp_sleep_get_wakeup_cause();
              // esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);

              if ( mode   == 0)                   {             M5Dial.Display.drawString("Start",        M5Dial.Display.width() / 2,           M5Dial.Display.height() / 2 - 40); }

              if ( mode   == 1)                   { mode   = 0; M5Dial.Display.drawString("Result",        M5Dial.Display.width() / 2,           M5Dial.Display.height() / 2 - 40);
                                                    switch(wakeup_reason)        {
                                                                                   case ESP_SLEEP_WAKEUP_EXT0 :     M5Dial.Display.drawString("Touch"  , M5Dial.Display.width() / 2,  M5Dial.Display.height() / 2 + 30); break;
                                                                                   case ESP_SLEEP_WAKEUP_EXT1 :     M5Dial.Display.drawString("Ext1"   , M5Dial.Display.width() / 2,  M5Dial.Display.height() / 2 + 30); break;
                                                                                   case ESP_SLEEP_WAKEUP_TIMER :    M5Dial.Display.drawString("timer"  , M5Dial.Display.width() / 2,  M5Dial.Display.height() / 2 + 30); break;
                                                                                   case ESP_SLEEP_WAKEUP_TOUCHPAD : M5Dial.Display.drawString("???"    , M5Dial.Display.width() / 2,  M5Dial.Display.height() / 2 + 30); break;
                                                                                   case ESP_SLEEP_WAKEUP_ULP :      M5Dial.Display.drawString("5"      , M5Dial.Display.width() / 2,  M5Dial.Display.height() / 2 + 30); break;
                                                                                   default :                        M5Dial.Display.drawString("button" , M5Dial.Display.width() / 2,  M5Dial.Display.height() / 2 + 30); break;
                                                                                 }
                                                    delay(5000);
                                                    M5Dial.Display.clear();
                                                  }
            }

void loop()
            {

              M5Dial.update();
              M5Dial.Display.drawString("Sleep tests",        M5Dial.Display.width() / 2,           M5Dial.Display.height() / 2 - 70);

              esp_sleep_wakeup_cause_t wakeup_reason;         wakeup_reason = esp_sleep_get_wakeup_cause();

              if ( mode   == 0)                   {             M5Dial.Display.drawString("Start",        M5Dial.Display.width() / 2,           M5Dial.Display.height() / 2 - 40); }

              if ( mode   == 1)                   { mode   = 0; M5Dial.Display.drawString("Result",        M5Dial.Display.width() / 2,           M5Dial.Display.height() / 2 - 40);
                                                    switch(wakeup_reason)        {
                                                                                   case ESP_SLEEP_WAKEUP_EXT0 :     M5Dial.Display.drawString("Touch"  , M5Dial.Display.width() / 2,  M5Dial.Display.height() / 2 + 30); break;
                                                                                   case ESP_SLEEP_WAKEUP_EXT1 :     M5Dial.Display.drawString("Ext1"   , M5Dial.Display.width() / 2,  M5Dial.Display.height() / 2 + 30); break;
                                                                                   case ESP_SLEEP_WAKEUP_TIMER :    M5Dial.Display.drawString("timer"  , M5Dial.Display.width() / 2,  M5Dial.Display.height() / 2 + 30); break;
                                                                                   case ESP_SLEEP_WAKEUP_TOUCHPAD : M5Dial.Display.drawString("???"    , M5Dial.Display.width() / 2,  M5Dial.Display.height() / 2 + 30); break;
                                                                                   case ESP_SLEEP_WAKEUP_ULP :      M5Dial.Display.drawString("5"      , M5Dial.Display.width() / 2,  M5Dial.Display.height() / 2 + 30); break;
                                                                                   default :                        M5Dial.Display.drawString("button" , M5Dial.Display.width() / 2,  M5Dial.Display.height() / 2 + 30); break;
                                                                                 }
                                                    delay(5000);
                                                    M5Dial.Display.clear();
                                                  }

//-------------- 1) To test light sleep initiate by the button and wakeup from the button
[... cut ...]
                                                  M5Dial.Display.drawString("Touch ",           M5Dial.Display.width() / 2,           M5Dial.Display.height() / 2 + 20);
                                                  delay(1000);
                                                  M5Dial.Display.clear();
                                                  esp_sleep_enable_ext0_wakeup(GPIO_NUM_14, LOW);
                                                  esp_deep_sleep_start();
                                             }

            }

Envoyé à partir de Courrier<https://go.microsoft.com/fwlink/?LinkId=550986> pour Windows

jsgagnon2020, 2024-02-10:
After more testing,

I found out that if you plug a battery in the M5Dial, the touchPad AND the button**  work for the deepsleep
(button** ONLY if the M5Dial as a battery as a power source)

Thank to all

jsgagnon2020, 2024-02-10:
Hi,

I have updated the full version of an Arduino M5dial deepsleep  program on Git

By any chance, do you know if there is some possibility of getting a status of the battery connected to the M5Dial?

In documentation, they appears to say NO??

Thank, JS

Envoyé à partir de Courrier<https://go.microsoft.com/fwlink/?LinkId=550986> pour Windows