m5stack/M5Dial #12: RFID Question.

kd8bxp, 2024-01-11:
From the documentation https://docs.m5stack.com/en/core/M5Dial
The Dial uses the WS1850S rfid chip, the same as the RFID2 Unit
https://docs.m5stack.com/en/unit/rfid2
I have no problem reading NFC512 ISO14443A Sticker tags with the RFID2 Unit, but the Dial will sometimes say UNSUPPORTED, or mostly it just refuses to read them at all.
IF it's using the same WS1850S chip, should it also read the same tags?
I used the demo code for the RFID2 Unit and the M5StickC-Plus without issue.
And the demo RFID example here in this repository for the Dial.
Both will read the credit card type MIFARE cards with out any problems.

I am very confused by this behavior thou, any suggestions - or maybe a different sketch/library to try.
Thanks.

shades66, 2024-01-18:
I'm not sure the Dial uses the WS1850S as the schematic references RC522 and the library in this repo (in src/utility) seems to reference the MFRC522 chipset 
https://www.nxp.com/docs/en/data-sheet/MFRC522.pdf

saying that the WS1850S does seem to have the same pinout so maybe a drop in replacement?
https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/products/core/M5Dial/WS1850S.PDF

anyway the demo doesn't seem to work with NTAG215 rfid cards, but can't see if that is something to do with the code only looking for mifare cards or whichever chip this has just doesn't like them !?!

bettse, 2024-02-17:
I was able to get NTAG215s to work: https://gitlab.com/bettse/m5dial_amiibo/
None of the NTAG215s I have are authentic (those are harder to find in general these days), and I noticed a lot more timeouts with some from "Timeskey" on Amazon vs a set I got on AliExpress a few years ago.

kd8bxp, 2024-02-23:
@bettse I tried that code and couldn't get it to read any of my NTAG215 - perhaps my tags are counterfeit, but the RFID II Unit reads them just fine, so I don't know.

This is what I have...
[image]
[image]
[image]

lucasromary, 2024-03-26:
Hello,

I have the same issue as kd8bxp.

I have Core2 and a RFID2 module reading a RFID Tag (ISO 14443-4) with this code (https://github.com/m5stack/M5Stack/tree/master/examples/Unit/RFID_RC522). No issue.

But with the M5Dial and the rfid example, the exact same card is not found. 
In the M5Dial -> src -> utility folder, i found a library that is quite similar to the library is used with the Core2.

My question is, is it a software issue ? Or is it hardware (RFID antenna is not powerfull enough, RFID IC is a little bit different from the RFID2 module) ?

miguelpineiroo, 2025-03-19:
Same here. Dial working with cards, not tags.
I have found the following Q&A in the official M5stack web:

Q1:
Dial cannot scan the RFID of my wireless key.
A:
Currently, Dial can only scan ID cards of standard bank card size, and other types of cards are not supported yet.

The answer suggest that NFC tags are not supported, only cards, regardless of the protocols/specs. It may be a signal strength issue related to the coil size in cards vs tags, but not sure