m5stack/M5Dial #13: problem with memory in Arduino IDE

gdivry, 2024-02-16:
I have a project that was running ok configured on 4 Mb and the 4Mb partition scheme 
(4Mb with SPIFF)
i added a few function and now it block because  saying i am using 133% of program memory

As it is an StampS3 supposed to have 8M 
i declared 8Mb in the tool menu ans the 8mb Partition scheme  but now the upload of code  always stop at 64% now  

how can i change the partition scheme on 4mb configuration ( i dont use spiff)  ?
or how to use the 8Mb conf ?

my project is stopped  :(

Sincerely 
Gery

gdivry, 2024-02-16:
to complete, i use the STAMPS board configuration because the M5stampS3 board config refuse to upload code

imliubo, 2024-02-18:
Hi @gdivry ,

Pin 9 is LCD backlight control pin, here is the [schematic](https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/products/core/M5Dial/Sch_M5Dial.pdf), 
[image]
Q1 in the picture is NMOS. When you set LCD_BL(Pin 9) to high level, the LCD backlight will be lit.

imliubo, 2024-03-19:
Closed for now, feel free to reopen.

closed 2024-03-19 by imliubo