m5stack/M5Dial #14: M5DIal Battery port

bettse, 2024-02-19:
What connector does the M5Dial's battery port use?  This would be the one inside, below the M5Stamp

[image]

I've seen speculation (https://old.reddit.com/r/M5Stack/comments/18ppxmr/what_kind_of_battery_plug_does_the_m5_dial_use/), but when I purchased JST-ZH (https://a.co/d/h6YO8UD) and JST-SH (https://a.co/d/5xiDnRJ), neither fit (one slightly too big, one slightly too small).  I haven't seen it documented (http://docs.m5stack.com/en/core/M5Dial) or on their forums.  I'm looking for an official, or tested answer, since all I've seen is speculation.

indus, 2024-02-22:
If JST-ZH (1.5mm) is to big and JST-SH (1mm) is to small is to small - i would guess it is a JST-GH (1.25mm) connector: https://en.wikipedia.org/wiki/JST_connector

The above is untested and only more speculation 😁 

But I'm quite confident as for example this panel also uses the JST-GH for battery: https://www.lilygo.cc/products/t-rgb (and to me it looks exactly the same)

bettse, 2024-02-24:
I could have made the same speculation, which is why I'm asking for tested, confirmed, or _official_ word on what it is before I move forward with further purchases.

wichjan, 2024-03-14:
If this is still of interest: The right connector is the JST-GH 1.25mm 2PIN
Tested and used by myself.
By the way: @M5Stack-Support: Your help here on github is needed!!! 
The M5Dial as a piece of hardware is great, but the documentation and the software-support really needs to be improved!!!

imliubo, 2024-03-19:
Sorry, we will add this to our document. Thanks for remind.

DevDianDankie, 2024-10-07:
Thank you @wichjan finally found a place that stock them. Could you please share what battery you use for the device? and how did you know what battery to use?

Is the voltage also between 6 and 36 for that connector?

wichjan, 2024-10-07:
Hi [DevDianDankie](https://github.com/DevDianDankie),
this port is **only** suitable for single cell Lithium-Ion/LithiumPolymer batteries (1S-LiIon/LiPo) with the nominal voltage of 3.7 V (max at full charge 4.2V), as the port is directly connected to a LiPo-charging circuit around the TP4057 chip. You can see this in the hardware schematics provided by m5Stack in the docs-section.
The connector is **not** capable for 6–36 V!! This will damage the hardware.
I am using a Lipo-battery with 500mAh capacity. You can find such everywhere from ebay to aliexpress, mouser, digikey etc.
They look like this: 
[image]

closed 2024-10-07 by bettse

bmatheny, 2024-10-20:
@wichjan my board has a "B+" printed on it. Should I assume that this is the battery polarity? If so it looks like the JST GH connector needs to be installed upside down (black on the B+ side).

wichjan, 2024-10-21:
Right. »B+« is the plus pole of the battery. There is no standard amongst the battery manufacturers, so you will find batteries with the red (+) and black (-) wires on either side of the JST connector.
So you may need to swap the red and black wires, so that the result looks like this:
[image]