m5stack/M5Dial #17: Internal i2c pullups

nikthefix, 2024-06-05:
Looking at the schematic, there don't appear to be any i2c bus pullup resistors for the shared Touch, RTC and RFID lines.
I'm experiencing corrupted data from the RTC. Could this be the cause of the problem or are the lines pulled to 3.3V elsewhere - in a way that I've missed?

Thanks

nik

felmue, 2024-06-05:
Hello @nikthefix 

checking with [gpio_dump_io_configuration](https://docs.espressif.com/projects/esp-idf/en/v5.2.2/esp32s3/api-reference/peripherals/gpio.html#_CPPv426gpio_dump_io_configurationP4FILE8uint64_t) it looks like GPIO11 and GPIO12 have (soft) pull-ups enabled. That said, I wasn't able to find where in the setup code the (soft) pull-ups are actually enabled.

Have you tried to lower the i2c speed?

Thanks
Felix

nikthefix, 2024-06-05:
Yes I've recommended reducing the i2c from 400 to 100kHz here:

In RTC8563_Class.hpp 
Set line 85 to RTC8563_Class(std::uint8_t i2c_addr = DEFAULT_ADDRESS, std::uint32_t freq = 100000,

I don't have the M5Dial product myself but have reproduced it on breadboard and am trying to help a colleague. 
I have his sketch working perfectly on the CoreS3.
The CoreS3 schematic shows the system I2C pullup resistors as expected.
The M5Dial schematic shows no pullups.
Internal pullups will not be sufficient in this case - based on my experience.

Thanks for responding.

nik

felmue, 2024-06-05:
Hello @nikthefix 

hmm, I wonder if that actually changes the frequency as `Wire1.getClock()` already always returns 100'000. E.g. at the very beginning of the program as well as after RTC IC interaction. (And again, I was not able to find in the code / libraries where the frequency for Wire1 gets set to 100'000.)

Maybe try to change the frequency with `Wire1.setClock(50000)` to even a lower value?

Thanks
Felix

esp32beans, 2024-06-06:
The rtc.ino example works on my M5Dial. Reading the touch screen works without problems.

nikthefix, 2024-06-06:
@esp32beans 

Yes the rtc.ino example works for me too but the sketch has very low i2c traffic and as @felmue pointed out, the internal weak pullups are employed and probably sufficient in this case. More frequent polling of the RTC chip at 400kHz seems to cause data corruption. Best case scenario is that the Lilygo schematic is wrong and that this problem is caused by something else entirely. 
But I have doubts.

nik

beniroquai, 2024-09-15:
Hey @nikthefix may I ask you how you fixed the error in your case? Did you add pullups ?

nikthefix, 2024-09-16:
@beniroquai I solved the problem on my breadboard reproduction of the M5Dial by adding modest 10K pullup resistors to the i2c lines. Everything worked as expected after that.

Are you experiencing problems with the M5Dial?

If I had the M5Dial hardware I would attempt to do the same as a mod. 
It's possible that M5stack have revised the board for later iterations. 

Worth checking out the latest schematic.

nik

beniroquai, 2024-09-16:
Thanks! I think I'm still stuck with the two different I2C "ports". It seems that changing the frequency of one also alters the other (?). I want to use the Dial as a slave. Changing wire.begin() as slave on 100kHz won't allow me to use the touch. If I initialize at 400kHz it works. Not sure how error prone this is as I use simple wires..

nikthefix, 2024-09-16:
@beniroquai I don't know which framework you are using but I would guess that to setup two i2c busses at different frequencies you would need to create separate objects - so wire.begin() and wire2.begin()...

Yes keep 400kHz for touch, RTC and RFID then use your wire2 as a separate entity for your slave comms.
There are examples of concurrent i2c buses included in the ESP32_Arduino package.

beniroquai, 2024-09-17:
I see. Thanks @nikthefix! I'll check the Wire2.begin() vs Wire.begin(). Did you happen to know where I can find an example for the M5Dial? Might be very generic anyway..

nikthefix, 2024-09-19:
@beniroquai I don't know any specific M5Dial examples but initializing a second i2c bus with different parameters on esp32-s3 is well documented on-line and in the Espressif literature. 

For Arduino IDE development checkout this tutorial:

https://randomnerdtutorials.com/esp32-i2c-communication-arduino-ide/

It's a great site for all sorts of solutions and I recon will show you how to do what you need in this case.

nik