m5stack/M5Dial #18: wifi not work when rfid is enable

studiosacchetti, 2024-07-12:
even a simple code like this can raise my problem

`//connect to WiFi
  Serial.printf("Connecting to %s ", ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println(" CONNECTED");
`

with this config setting everything is ok

`  M5Dial.begin(cfg, false, false);`

but if i enable rfid 

`  M5Dial.begin(cfg, false, true);`

m5dial can connect only once to my wifi network, i have a fritzbox router

any idea?

Gitshaoxiang, 2024-08-02:
we have tested the RFID function with wifi, and also tested reconnecting multiple times, is problem. Hi, could you provide the full code you're using?

studiosacchetti, 2024-08-04:
```
/**
   @file rfid.ino
   @author SeanKwok (shaoxiang@m5stack.com)
   @brief M5Dial RFID Test
   @version 0.2
   @date 2023-10-18

   @Hardwares: M5Dial
   @Platform Version: Arduino M5Stack Board Manager v2.0.7
   @Dependent Library:
   M5GFX: https://github.com/m5stack/M5GFX
   M5Unified: https://github.com/m5stack/M5Unified
*/

#include "M5Dial.h"
#include <HTTPClient.h>
#include <ArduinoJson.h>

//Your Domain name with URL path or IP address with path
const char* serverName = "http://arduinojson.org/example.json";

const char* ssid       = "xxxx";
const char* password   = "yyyyy";
String sensorReadings;

void setup() {

  auto cfg = M5.config();
  cfg.serial_baudrate = 115200;
  M5Dial.begin(cfg, false, true);
  M5Dial.Display.setTextColor(GREEN);
  M5Dial.Display.setTextDatum(middle_center);
  M5Dial.Display.setTextFont(&fonts::Orbitron_Light_32);
  M5Dial.Display.setTextSize(1);

  M5Dial.Display.drawString("RFID Card", M5Dial.Display.width() / 2, M5Dial.Display.height() / 2);

  WiFi.begin(ssid, password);
  Serial.println("Connecting");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to WiFi network with IP Address: ");
  Serial.println(WiFi.localIP());

  //contrrollare il server

}

void loop() {
  if (M5Dial.Rfid.PICC_IsNewCardPresent() && M5Dial.Rfid.PICC_ReadCardSerial()) {

    M5Dial.Display.clear();

    Serial.print(F("PICC type: "));
    uint8_t piccType = M5Dial.Rfid.PICC_GetType(M5Dial.Rfid.uid.sak);
    Serial.println(M5Dial.Rfid.PICC_GetTypeName(piccType));

    // Check is the PICC of Classic MIFARE type
    if (piccType != MFRC522::PICC_TYPE_MIFARE_MINI &&  piccType != MFRC522::PICC_TYPE_MIFARE_1K && piccType != MFRC522::PICC_TYPE_MIFARE_4K) {
      Serial.println(F("Your tag is not of type MIFARE Classic."));
      M5Dial.Display.drawString("not support", M5Dial.Display.width() / 2, M5Dial.Display.height() / 2);
      //return;
    }
    String uid = "";

    for (byte i = 0; i < M5Dial.Rfid.uid.size;  i++) {
      Serial.printf("%02X ", M5Dial.Rfid.uid.uidByte[i]);
      uid += String(M5Dial.Rfid.uid.uidByte[i], HEX);
    }

    sensorReadings = httpGETRequest(serverName);
    Serial.println(sensorReadings);
    // Parse response
    //const size_t capacity = JSON_OBJECT_SIZE(3) + JSON_ARRAY_SIZE(2) + 60;
    DynamicJsonDocument doc(2048);
    deserializeJson(doc, sensorReadings);

    /*if (error) {
      Serial.print(F("deserializeJson() failed: "));
      Serial.println(error.f_str());
      
      //client.stop();
      return;
    }*/
    Serial.println(doc["time"].as<long>());

    //JSONVar myObject = JSON.parse(sensorReadings);

    // JSON.typeof(jsonVar) can be used to get the type of the var
    /*if (JSON.typeof(myObject) == "undefined") {
      Serial.println("Parsing input failed!");
      return;
    }*/

    M5Dial.Speaker.tone(8000, 20);
    M5Dial.Display.drawString(M5Dial.Rfid.PICC_GetTypeName(piccType), M5Dial.Display.width() / 2, M5Dial.Display.height() / 2 - 30);
    M5Dial.Display.drawString("card id:", M5Dial.Display.width() / 2, M5Dial.Display.height() / 2);
    M5Dial.Display.drawString(uid, M5Dial.Display.width() / 2, M5Dial.Display.height() / 2 + 30);
    Serial.println();
  }
}

String httpGETRequest(const char* serverName) {
  HTTPClient http;

  // Your IP address with path or Domain name with URL path
  http.begin(serverName);

  // If you need Node-RED/server authentication, insert user and password below
  //http.setAuthorization("REPLACE_WITH_SERVER_USERNAME", "REPLACE_WITH_SERVER_PASSWORD");

  // Send HTTP POST request
  int httpResponseCode = http.GET();

  String payload = "{}";

  if (httpResponseCode > 0) {
    Serial.print("HTTP Response code: ");
    Serial.println(httpResponseCode);
    payload = http.getString();
  } else {
    Serial.print("Error code: ");
    Serial.println(httpResponseCode);
  }
  // Free resources
  http.end();

  return payload;
}
```

you may notice that wifi connection is not functionig and also serial output.

andy

studiosacchetti, 2024-08-04:
`/**
   @file rfid.ino
   @author SeanKwok (shaoxiang@m5stack.com)
   @brief M5Dial RFID Test
   @version 0.2
   @date 2023-10-18

   @Hardwares: M5Dial
   @Platform Version: Arduino M5Stack Board Manager v2.0.7
   @Dependent Library:
   M5GFX: https://github.com/m5stack/M5GFX
   M5Unified: https://github.com/m5stack/M5Unified
*/

#include "M5Dial.h"
#include <HTTPClient.h>
#include <ArduinoJson.h>

//Your Domain name with URL path or IP address with path
const char* serverName = "http://arduinojson.org/example.json";

const char* ssid       = "XXXX";
const char* password   = "YYYY";
String sensorReadings;

void setup() {

  auto cfg = M5.config();
  cfg.serial_baudrate = 115200;
  M5Dial.begin(cfg, false, true);
  M5Dial.Display.setTextColor(GREEN);
  M5Dial.Display.setTextDatum(middle_center);
  M5Dial.Display.setTextFont(&fonts::Orbitron_Light_32);
  M5Dial.Display.setTextSize(1);

  M5Dial.Display.drawString("RFID Card", M5Dial.Display.width() / 2, M5Dial.Display.height() / 2);

  WiFi.begin(ssid, password);
  Serial.println("Connecting");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to WiFi network with IP Address: ");
  Serial.println(WiFi.localIP());

  //contrrollare il server

}

void loop() {
  if (M5Dial.Rfid.PICC_IsNewCardPresent() && M5Dial.Rfid.PICC_ReadCardSerial()) {

    M5Dial.Display.clear();

    Serial.print(F("PICC type: "));
    uint8_t piccType = M5Dial.Rfid.PICC_GetType(M5Dial.Rfid.uid.sak);
    Serial.println(M5Dial.Rfid.PICC_GetTypeName(piccType));

    // Check is the PICC of Classic MIFARE type
    if (piccType != MFRC522::PICC_TYPE_MIFARE_MINI &&  piccType != MFRC522::PICC_TYPE_MIFARE_1K && piccType != MFRC522::PICC_TYPE_MIFARE_4K) {
      Serial.println(F("Your tag is not of type MIFARE Classic."));
      M5Dial.Display.drawString("not support", M5Dial.Display.width() / 2, M5Dial.Display.height() / 2);
      //return;
    }
    String uid = "";

    for (byte i = 0; i < M5Dial.Rfid.uid.size;  i++) {
      Serial.printf("%02X ", M5Dial.Rfid.uid.uidByte[i]);
      uid += String(M5Dial.Rfid.uid.uidByte[i], HEX);
    }

    sensorReadings = httpGETRequest(serverName);
    Serial.println(sensorReadings);
    // Parse response
    //const size_t capacity = JSON_OBJECT_SIZE(3) + JSON_ARRAY_SIZE(2) + 60;
    DynamicJsonDocument doc(2048);
    deserializeJson(doc, sensorReadings);

    /*if (error) {
      Serial.print(F("deserializeJson() failed: "));
      Serial.println(error.f_str());
      
      //client.stop();
      return;
    }*/
    Serial.println(doc["time"].as<long>());

    //JSONVar myObject = JSON.parse(sensorReadings);

    // JSON.typeof(jsonVar) can be used to get the type of the var
    /*if (JSON.typeof(myObject) == "undefined") {
      Serial.println("Parsing input failed!");
      return;
    }*/

    M5Dial.Speaker.tone(8000, 20);
    M5Dial.Display.drawString(M5Dial.Rfid.PICC_GetTypeName(piccType), M5Dial.Display.width() / 2, M5Dial.Display.height() / 2 - 30);
    M5Dial.Display.drawString("card id:", M5Dial.Display.width() / 2, M5Dial.Display.height() / 2);
    M5Dial.Display.drawString(uid, M5Dial.Display.width() / 2, M5Dial.Display.height() / 2 + 30);
    Serial.println();
  }
}

String httpGETRequest(const char* serverName) {
  HTTPClient http;

  // Your IP address with path or Domain name with URL path
  http.begin(serverName);

  // If you need Node-RED/server authentication, insert user and password below
  //http.setAuthorization("REPLACE_WITH_SERVER_USERNAME", "REPLACE_WITH_SERVER_PASSWORD");

  // Send HTTP POST request
  int httpResponseCode = http.GET();

  String payload = "{}";

  if (httpResponseCode > 0) {
    Serial.print("HTTP Response code: ");
    Serial.println(httpResponseCode);
    payload = http.getString();
  } else {
    Serial.print("Error code: ");
    Serial.println(httpResponseCode);
  }
  // Free resources
  http.end();

  return payload;
}`

you may notice some random weirdness like wifi not connecting or empty serial output

esp32beans, 2024-08-04:
The problem might be due to WiFi + RFID drawing too much current. WiFi draws lots of current when scanning for access points (that is when disconnected).Try connecting to WiFi with RFID off. After WiFi connected, turn on RFID. If disconnected, turn RFID off before reconnecting WiFi.

Or use a power supply capable of supplying 5V 1A or more. USB ports on computer usually can only supply 0.5A. Some powered USB hubs can supply more than 0.5A.

studiosacchetti, 2024-08-04:
true. i will try
thx

studiosacchetti, 2024-08-05:
hello
i have tried with several power supply but this doesen't fix my issue. ti seems that the only way to make it work is usign this ```
setup

```
MFRC522 mfrc522(0x28);  // Create MFRC522 instance.  创建MFRC522实例

void setup() {
  /*auto cfg = M5.config();
    cfg.serial_baudrate = 115200;
    M5Dial.begin(cfg, false, false);*/

  auto cfg = M5.config();
  cfg.serial_baudrate = 115200;
  M5Dial.begin(cfg, false, false);
 

.
.
.
 
 
 
  mfrc522.PCD_Init();  // Init MFRC522.  初始化 MFRC522

 
}

```

studiosacchetti, 2024-08-05:
hello i have tried many configuration and different power supply, but i did not solve my issue. it doesen't work
using this setup partially fix the problem

```
  auto cfg = M5.config();
  cfg.serial_baudrate = 115200;
  M5Dial.begin(cfg, false, false);

 mfrc522.PCD_Init();  // Init MFRC522.  初始化 MFRC522

```

any ideas? i can send by email the full project if u want

esp32beans, 2024-08-06:
Sorry, I do not have time to debug.

Turning down or off the display backlight when connecting WiFi might help. The backlight uses lots of current and I think defaults to maximum brightness. After connecting, turn the brightness up as desired.

```
M5Device.Display.setBrightness(M5_BRIGHTNESS);
```

studiosacchetti, 2024-08-13:
ok will try soon. Is there any power supply that u may recommend to solve this issue? better usb or not?. i am  a software developer not  a electronic engineer
regards

PaulskPt, 2024-11-27:
I created a [project](https://github.com/PaulskPt/M5Dial_Timezones_data_to_Google/blob/main/docs/Monitor_output.txt) using Arduino IDE v2.3.3. This project uses the RFID and WiFi functionalities of the M5Dial. Power USB or external power from a Raspberry Pi, model P4421, AC adapter giving 8 Volts @ 6 Amps. I have no problems establishing WiFi connection while also using RFID to identfy an RFID TAG card.