m5stack/M5Dial #19: Wire.h: No such file or directory

ktetsuo, 2024-08-25:
I get a build error when using platform.io.

```
In file included from .pio/libdeps/m5stack-stamps3/M5Dial/src/M5Dial.h:7,
                 from .pio/libdeps/m5stack-stamps3/M5Dial/src/M5Dial.cpp:1:
.pio/libdeps/m5stack-stamps3/M5Dial/src/utility/MFRC522.h:83:10: fatal error: Wire.h: No such file or directory

 #include <Wire.h>
          ^~~~~~~~
compilation terminated.
In file included from .pio/libdeps/m5stack-stamps3/M5Dial/src/utility/MFRC522.cpp:10:
.pio/libdeps/m5stack-stamps3/M5Dial/src/utility/MFRC522.h:83:10: fatal error: Wire.h: No such file or directory
```

platformio.ini:
```
[env:m5stack-stamps3]
platform = espressif32
board = m5stack-stamps3
framework = arduino
lib_deps = 
	m5stack/M5Dial@^1.0.2
	m5stack/M5Unified@^0.1.16
```

ktetsuo, 2024-08-25:
Commenting out `#include <Wire.h>`, then the build succeeded.
`Wire.h` seems unnecessary.

src\utility\MFRC522.h
```diff
#ifndef MFRC522_h
#define MFRC522_h

#include "M5Unified.h"
#include <Arduino.h>
- #include <Wire.h>
+ //#include <Wire.h>
```

src\utility\MFRC522.cpp
```diff
#include <Arduino.h>
#include "MFRC522.h"
- #include <Wire.h>
+ //#include <Wire.h>

using namespace m5;
```

Tinyu-Zhao, 2024-08-27:
I can't reproduce, can you prove more information.
[image]

ktetsuo, 2024-08-31:
I solved by myself.

It is neccesary to add `Wire` to `lib_deps`.
https://community.platformio.org/t/wire-h-not-found-platformio-latest-version/2046

After adding `lib_deps = Wire`, the build succeeded.

```diff
[env:m5stack-stamps3]
platform = espressif32
board = m5stack-stamps3
framework = arduino
lib_deps = 
	m5stack/M5Dial@^1.0.2
	m5stack/M5Unified@^0.1.16
+	Wire
```

closed 2024-08-31 by ktetsuo